CrealityOfficial/E3-Free-runs-Silent-Motherboard #1: STM32F401 Chip & Other Changes

lokiagent, 2023-02-21:
I received this board today and started installing it. I've run in to several issues so far. The first is that the board is a CR4NT220801C10 with a STM32F401 processor rather than a CR4NT2206C10 with an STM32F103. I've compiled the right firmware after trial and error, but I ran in to an issue with the bed heater, using the pins for the CR4NT2206C10 works for everything except the bed, which blows the fuse as soon as you signal to heat the bed. Could you please provide the proper pin out locations for this variation of the board?

lokiagent, 2023-02-21:
After looking around more it looks like PB10 is the correct pin for the heated bed, but I'm guessing a mosfet blew out with the fuse because now the bed is heating any time the board has power.

closed 2023-02-21 by lokiagent

tflucke, 2024-09-22:
Thank you, LokiAgent!  I received a Creality 4.2.7 STM32F401 motherboard yesterday and have spent nearly all day trying to figure out why my heatbed was running for seemingly no reason.  My config has PB10 running to the display which was causing the heat bed to always be on.