CrealityOfficial/E3-Free-runs-Silent-Motherboard #3: E3 Runs Free BL Touch Firmware needed

Jillybowl, 2023-04-18:
I have been trying to find or get the firmware for this new board for the BL Touch, E3 Runs-Free on an Ender 3 Pro.  I am ready to put the old board in and go back to the way it was due to issue I have had.   Please help

dpjpandone, 2023-04-18:
congratulations you found it!

dpjpandone, 2023-04-18:
https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/tree/main/firmware/Marlin

Jillybowl, 2023-04-19:
how do you enable bl touch in this firmware, it is what is loaded not he printer now but no option for bedeviling

dpjpandone, 2023-04-19:
If you compare the connfig.h from the ender3-E branch you can copy the settings ad recompile. the main things are define:
bltouch
bilinearLeveling

and set your nozzle to probe offsets to -43, -10

Jillybowl, 2023-04-20:
Do you have instructions for this, I have no idea how to do any of that.  Can't you just compile one for everyone to download with blouch like all the other have available

dpjpandone, 2023-04-20:
I don't work here. I'm just a regular guy like you. I googled "how to compile marlin" and this is the first thing that popped up:

https://all3dp.com/2/marlin-firmware-how-to-edit-marlin-firmware/

Jillybowl, 2023-04-25:
Im sending it back, I have no idea where to eve start to get the correct files to do this, I installed Visual Studio, platform, marlin builder.  No idea where or how to download the correct source files that are needed, I only find examples that I can't download or just the .bin file

dpjpandone, 2023-04-26:
The source code is here: https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/tree/main/firmware/Marlin

you have to click on "code" and download zip. Then you explore that folder in vscode and compile with marlin auto build

Jillybowl, 2023-04-29:
Returned the E3 Runs Free board and got a 4.2.7, opened the box and 38  minutes later I was printing.  Thank you amazon reseller as creality said they don't have them any more

closed 2023-04-29 by Jillybowl