CrealityOfficial/E3-Free-runs-Silent-Motherboard #4: Does't work for CR4NT220801C10

ElasticSky, 2023-05-04:
When compiling for CR4NT220801C10 version of the E3 Free-Run board with BLTouch support, BLTouch will not deploy.

Expected behavior would be that there was support for this board in the Marlin firmware.

Jillybowl, 2023-05-04:
Return the board, they have no support for it what so ever

ElasticSky, 2023-05-10:
> Return the board, they have no support for it what so ever

Thanks - that was the feeling I got as well. Even Creality support don't have any information on the board. Will do as you suggest.

Eutectoid88, 2023-05-24:
Try this:
Working with Ender 3 V2 Neo; SonicPad and SpritePro extruder!

[bltouch]
sensor_pin: PB1
control_pin: PB0
pin_move_time: 0.5
pin_up_reports_not_triggered: False
pin_up_touch_mode_reports_triggered: True
x_offset: -36.5
y_offset: -40
#z_offset: -5
speed:3

[safe_z_home]
home_xy_position: 145,155
speed: 150
z_hop: 10
z_hop_speed: 10

[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 30, 30
mesh_max: 190, 190
probe_count: 3,3
algorithm: bicubic
#fade_start: 1
#fade_end: 10
#fade_target: 0

Catvari, 2023-05-29:
#define Z_MIN_PROBE_PIN PB1

It correct the problem for me

Aaron1073, 2023-06-16:
This is how I got the **_E3 Free-Runs_** working working on a Ender-3 Pro with a CR-Touch / BLTouch, I am still testing, it seems to have homed right and has successfully completed and bed Leveling. I have yet to set the offset and do a print or test the bed / hot-end heaters. I will continue to update this post as I work on this over the weekend.

I find it really sad that Creality3D could not help with this, to them it may not be allot of money but after ( 2 ) Ender-3, ( 2 ) Ender-3 Pro and a Ender-3 V2... close to 3000$ spent including upgrades, I may consider going to another brand.

If you use this it is at your own risk, I was willing to risk my board to try and get it working

In **_"\Marlin\src\pins\stm32f4\pins_CREALITY_CR4NTxxC10.h"_** I made the following changes
```
#define Z_STOP_PIN         PB0  // BLTouch IN PIN  原理图TOUCH的管脚已经变-----zy
#define SERVO0_PIN         PB1  // BLTouch PWM-OUT PIN  原理图TOUCH的管脚已经变-----zy
```
change to 
```
#define Z_STOP_PIN         PB1  // BLTouch IN PIN  原理图TOUCH的管脚已经变-----zy
#define SERVO0_PIN         PB0  // BLTouch PWM-OUT PIN  原理图TOUCH的管脚已经变-----zy
```

`#define Z_MIN_PROBE_PIN                   PB0   // BLTouch IN`

changed to

`#define Z_MIN_PROBE_PIN                   PB1   // BLTouch IN`

eshvatskyi, 2023-06-29:
> Try this: Working with Ender 3 V2 Neo; SonicPad and SpritePro extruder!
>
> [bltouch] sensor_pin: PB1 control_pin: PB0 pin_move_time: 0.5 pin_up_reports_not_triggered: False [...]

Can you post a whole config? I'm looking for an config for my ender 3 v2 new with sprite for this new board

ElasticSky, 2023-07-06:
@Eutectoid88 and @Catvari - legends. Both of you should take over support from Creality :)

I thought I had done this before but modified pins_CREALITY_CR4NTxxC10.h file as below and its working:

```
#ifdef BLTOUCH
  #define Z_STOP_PIN         PB1  // BLTouch IN PIN  原理图TOUCH的管脚已经变-----zy
  #define SERVO0_PIN         PB0  // BLTouch PWM-OUT PIN  原理图TOUCH的管脚已经变-----zy
  #define Z_STOP_PIN_NADD    PA7   //Added z-axis limit switch  rock_20210816
#else
  #define Z_STOP_PIN         PA7 //Z轴限位开关
#endif

//#define one (c14 || a15)

#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN                   PB1   // BLTouch IN
#endif
```

Obviously needed to set this in the configuration.h file as well given I'm using the 5-pin cable but that's a given:

```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING
```

Cheers guys so very very much.

ElasticSky, 2023-07-06:
@Aaron1073 - to remove the "TMC Connection Error" on the board in the middle is a jumper (J2). Either remove or set to MS-GND. This removed the error for me.

dannegomez, 2023-07-08:
> @Eutectoid88, @Aaron1073 and @Catvari - legends. All of you should take over support from Creality :)
> 
> I thought I had done this before but modified pins_CREALITY_CR4NTxxC10.h file as below and its working:
> 
> ```
> #ifdef BLTOUCH
>   #define Z_STOP_PIN         PB1  // BLTouch IN PIN  原理图TOUCH的管脚已经变-----zy
>   #define SERVO0_PIN         PB0  // BLTouch PWM-OUT PIN  原理图TOUCH的管脚已经变-----zy
>   #define Z_STOP_PIN_NADD    PA7   //Added z-axis limit switch  rock_20210816
> #else
>   #define Z_STOP_PIN         PA7 //Z轴限位开关
> #endif
> 
> //#define one (c14 || a15)
> 
> #ifndef Z_MIN_PROBE_PIN
>   #define Z_MIN_PROBE_PIN                   PB1   // BLTouch IN
> #endif
> ```

Awesome! big thanks @Eutectoid88 @Aaron1073 @Catvari and @ElasticSky! This change of pins fixed my issue with the CRTouch and the CR4NT220801C10 board.

Aaron1073, 2023-07-10:
> 

do you know what the jumper is for by chance? not that I am not going to try. just looking to understand better

Tried to print at 150mm/s but have been getting mixed results

The-sax-man, 2023-08-25:
Here is a firmware version for the cr touch for those who want it not sure if it will still work after being unzipped I haven't checked
[firmwhere.zip](https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/files/12436207/firmwhere.zip)

The-sax-man, 2023-08-25:
forget to but the bin file into a folder on the sd card called "STM32F4_UPDATE" so it will flash the printer

junkjunker, 2023-11-27:
@Aaron1073  What would this look like for Klipper?  I am using a KlackEnder variant (simple switch, not BL Touch or CR Touch) and can't get this board to place nice with what I thought would be a simple install.

junkjunker, 2023-11-29:
> Here is a firmware version for the cr touch for those who want it not sure if it will still work after being unzipped I haven't checked don't forget to but the bin file into a folder on the sd card called "STM32F4_UPDATE" so it will flash the printer
> 
> FOR ENDER 3 please don't try and use this on a different printer :)
> 
> [firmwhere.zip](https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/files/12436207/firmwhere.zip)

When I use this firmware on an Ender 3 Pro with Klipper, it just hangs:
"Klipper reports: STARTUP
Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments."

Then it times out with:
"Klipper reports: ERROR
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer"

The only firmware I've been able to get to run is the one supplied by Creality (Klipper_CR4NTxxC10.bin), but then the BL Touch/probe don't respond properly.  Is there a way I can edit that file to make it work?

THANKS

eugenystrigulin, 2024-01-14:
> In **_"\Marlin\src\pins\stm32f4\pins_CREALITY_CR4NTxxC10.h"_** I made the following changes
> 
> ```
> #define Z_STOP_PIN         PB0  // BLTouch IN PIN  原理图TOUCH的管脚已经变-----zy
> #define SERVO0_PIN         PB1  // BLTouch PWM-OUT PIN  原理图TOUCH的管脚已经变-----zy
> ```
> 
> change to
> 
> ```
> #define Z_STOP_PIN         PB1  // BLTouch IN PIN  原理图TOUCH的管脚已经变-----zy
> #define SERVO0_PIN         PB0  // BLTouch PWM-OUT PIN  原理图TOUCH的管脚已经变-----zy
> ```
> 
> then a few lines down
> 
> `#define Z_MIN_PROBE_PIN PB0 // BLTouch IN`
> 
> changed to
> 
> `#define Z_MIN_PROBE_PIN PB1 // BLTouch IN`

Thank U its work

donruan34, 2024-03-28:
> Aqui está uma versão de firmware para o cr touch para quem quiser, não tenho certeza se ainda funcionará depois de descompactado, não verifiquei, não se esqueça, mas o arquivo bin em uma pasta no cartão SD chamada "STM32F4_UPDATE" então ele piscará a impressora
> 
> PARA ENDER 3, não tente usar isso em uma impressora diferente :)
> 
> [firmwhere.zip](https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/files/12436207/firmwhere.zip)

Until I was able to install bltouch with the file you sent I was very happy but when I go to do a home it gives error m999 I saw that the z doesn't go down but you can see that...

donruan34, 2024-03-28:
Until I was able to install bltouch with the file you sent I was very happy but when I go to do a home it gives error m999 I saw that the z doesn't go down but you can see that...

Aaron1073, 2024-04-01:
sorry it has been a bit, I am out right now but when I get home to my computer I will upload my source for ppl to try out for the ppl trying to get it to work I can not be of much help I do not own one

Aaron1073, 2024-04-01:
> For Klipper, using the stock firmware from Creality ([Klipper_CR4NTxxC10.bin](https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/blob/main/firmware/Klipper/Klipper_CR4NTxxC10.bin), and @Eutectoid88 's [suggested pin assignments](https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/issues/4#issuecomment-1561937489), I was able to get this board to recognize my Klicky/KlackEnder-type probe. My probe's switch just connects to the BLTouch connector only using the "G" and "OUT" pins, and uses the standard end stop switches (including Z stop - I didn't disconnect that).
> 
> Here's how I changed the printer.cfg file for my Ender 3 Pro:
> 
> [stepper_x] #endstop_pin: !PA5 endstop_pin: PA5 # remove "!"
> 
> [stepper_y] #endstop_pin: !PA6 endstop_pin: PA6 # remove "!"
> 
> [stepper_z] #endstop_pin: !PA7 endstop_pin: PA7 # remove "!"
> 
> [probe] #pin: PC14 pin: ^PB1 #Probe-Stop Connection on Creality E3 Free-runs TMC2209 Open Source Silent Motherboard

Sorry I could not help with this I have never tried Klipper, I don't have allot of time to play. Work takes allot of my time and I only have the odd weekend to play, although I will try your comfig out

Anibal133, 2024-08-24:
Hello, if you have the TMC2209 firmware for ender 5 pro with frabica components, please send it to me, I tried to make one and my printout was unusable and with the screen completely blue, if anyone has the firmware send it and if it is possible with klipper

msw012, 2024-09-27:
Hi all,

OK. I'm also one of the poor schmucks who bought a Creality E3 Free-runs mother board thinking it would work with my Ender 5. I feel like I've been promised the world and then just given an atlas. That said, I've recently tried building Marlin myself and had some success on an Ender 5. The only thing not working so far is the nozzle temperature. The stepper control works (very smooth and quite), the BL Touch seems to work. 

Here's some background :
  1. Bought the board about 12 months ago only to discover that there's no firmware support.
  2. Contacted Creality by email for assistance eg. schematics to create configurations. After a bit of back and forth got nowhere. Basically the "computer says no".
  3. Decided to leave it for a bit and see if support leaks out over time (in spite of Creality's best efforts no doubt).
  4. Recently stumbled on some Marlin configuration files for the Free-runs and the Ender 5 at : https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5/Creality%20free-runs
  5. Worked out how to build Marlin under Linux so can create firmware and test it.

Here's some observations and speculations (guesses) :
 1. There seems to be two numbers associated with the Free-runs board ie. CR4NT220601C10 & CR4NT220801C10. A web search for CR4NT220601C10 produces virtually no results but for the CR4NT220801C10 quite a few.
 2. The CR4NT220601C10 uses the STM32F103RET6 MCU (shown in schematics).
 3. The CR4NT220801C10 uses the STM32F403RET6 MCU (fair sure of this).
 4. Perhaps the CR4NT220601C10 was an early version of the Free-runs board and CR4NT220801C10 is an update.
 5. The only schematics available are for the CR4NT220601C10 board and that's what the Marlin configuration files are developed for.
 6. The two Free-runs boards differ slightly. Ie. the MCU are almost pin for pin compatible but not quite.

Has anyone been able to get their hands on the CR4NT220801C10 schematics? I've asked Creality but to no avail. Might try again now because there seems to be some light at the end of the tunnel.

Any comments or suggestions are welcome.

msw012, 2024-10-19:
Hi all,

Good news. I've managed to build the Marlin firmware (version bugfix-2.1.x) and load it onto my Creality E3 Free-runs board which is installed in my Ender 5. Everything seems to work though I haven't printed anything yet. Need to setup the BLTouch before I can attempt a print. Will keep you posted.

msw012, 2024-10-25:
Even better news. Have tested the Marlin firmware on the Creality E3 Free-runs board in my Ender 5 with a CR Touch and it all works fine. The firmware used is the Marlin bugfix-v2.1.x branch, the other branches don't work with the Creality E3 Free-runs board yet. I'm building the Marlin firmware on Archlinux via the command line which turns out to be pretty straight forward. What follows is a excerpt from my personal howto on 3D printing :

Configure and build Marlin on Arch Linux as follows :

  a. Install build tool Platform-IO :

       $ yay -S platformio-core

  b. Download the Marlin sources file from : https://marlinfw.org/meta/download/ to ~/Downloads/.
     Eg. the file : Marlin-2.1.x.zip (the only release that works with E3 Free-runs board at present).

  c. Download the configuration files from : https://marlinfw.org/meta/download/ to ~/Downloads/.
     Ie. the files : Configuration.h, Configuration_adv.h, _Bootscreen.h and _Statusscreen.h.
     There are many configurations for all the various board and printers. For the Creality Ender 5
     and the Creality E3 Free-runs board the configuration files can found at :  
       https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5/Creality%20free-runs

  d. Unzip the Marlin sources in an appropriate location and move the configuration files :

       $ mv ~/Downloads/Marlin-bugfix-2.1.x.zip ~/libs/marlin/ender-5/
       $ cd ~/libs/marlin/ender-5/
       $ unzip Marlin-bugfix-2.1.x.zip
       $ mv Marlin-bugfix-2.1.x/ bugfix-v2.1.x/
       $ mv ~/Downloads/*.h ./bugfix-v2.1.x/Marlin/

  e. Modify the file ~/libs/Marlin/bugfix-v2.1.x/Marlin/Configuration.h by editing the following
     lines (line numbers correct as of 2024-10-20) :
  
       Line 1428 : #define BLTOUCH                                            -> Uncomment line
       Line 1604 : #define NOZZLE_TO_PROBE_OFFSET { -42, 5, 0}                -> Set parameter values
       Line 1613 : #define PROBING_MARGIN 25                                  -> Set parameter value
       Line 2069 : #define AUTO_BED_LEVELING_BILINEAR                         -> Uncomment line
       Line 2084 : #define RESTORE_LEVELING_AFTER_G28                         -> Uncomment line
       Line 2090 : #define PREHEAT_BEFORE_LEVELING                            -> Uncomment line
       Line 2092 : #define LEVELING_NOZZLE_TEMP 150                           -> Set parameter value
       Line 2152 : #define GRID_MAX_POINTS_X 5                                -> Set parameter value
       Line 2162 : #define EXTRAPOLATE_BEYOND_GRID                            -> Uncomment line
       Line 2168 : #define ABL_BILINEAR_SUBDIVISION                           -> Uncomment line
       Line 2238 : #define LCD_BED_LEVELING                                   -> Uncomment line
       Line 2305 : #define Z_SAFE_HOMING                                      -> Uncomment line

     Make the following modifications if using OctoPrint :

       Line 2467 : #define NOZZLE_PARK_FEATURE                  -> Uncomment line

  f. Modify the file ~/libs/Marlin/bugfix-v2.1.x/Marlin/Configuration_adv.h by editing the following
     lines (line numbers correct as of 2024-10-20) :

       Line 1510 : #define PROBE_OFFSET_WIZARD                                -> Uncomment line
       Line 1517 : #define PROBE_OFFSET_WIZARD_START_Z -4.0                   -> Uncomment line
       Line 1520 : #define PROBE_OFFSET_WIZARD_XY_POS { X_CENTER, Y_CENTER }  -> Uncomment line
       Line 1817 : #define SDCARD_SORT_ALPHA                                  -> Uncomment line

     Make the following modifications if using OctoPrint :

       Line 1678 : #define SET_PROGRESS_MANUALLY                              -> Uncomment line
       Line 1683 : #define M73_REPORT                                         -> Uncomment line
       Line 1693 : #define SHOW_REMAINING_TIME                                -> Uncomment line
       Line 1700 : #define LCD_PROGRESS_BAR                                   -> Uncomment line
       Line 1837 : #define LONG_FILENAME_HOST_SUPPORT                         -> Uncomment line
       Line 1857 : #define AUTO_REPORT_SD_STATUS                              -> Uncomment line
       Line 2521 : #define ARC_SUPPORT                                        -> Uncomment line
       Line 2626 : #define BLOCK_BUFFER_SIZE 64                               -> Set parameter value
       Line 2635 : #define BUFSIZE 32                                         -> Set parameter value
       Line 2644 : #define TX_BUFFER_SIZE 32                                  -> Set parameter value
       Line 2650 : #define RX_BUFFER_SIZE 2048                                -> Set parameter value
       Line 2682 : #define EMERGENCY_PARSER                                   -> Uncomment line
       Line 2713 : #define ADVANCED_OK                                        -> Uncomment line
       Line 2915 : #define ADVANCED_PAUSE_FEATURE                             -> Uncomment line
       Line 3924 : #define M114_DETAIL                                        -> Uncomment line
[... 42 lines cut ...]
  i. Build the Marlin binary (possible target environments STM32F401RE_freeruns,
     STM32F401RE_freeruns_jlink or STM32F401RE_freeruns_stlink) : 

       $ cd ~/libs/Marlin/bugfix-v2.1.x
       $ platformio run -e STM32F401RE_freeruns

[... cut ...]
  a. Create a micro SD card formatted with FAT32.

  b. Create a directory on the SD card : STM32F4_UPDATE

  c. Copy the binary file into the directory created in the last step of the previous section.
     Note : The name of the firmware file must differ from the previous file name as the printer
            remembers the last file name.

  d. Insert the micro SD card into the 3D printer.

  e. Powerup the 3D printer and the new firmware will automatically be detected and programmed into
     the mother board flash memory.

githubranzo, 2026-09-04:
Dear all,

I would like your help if possible, namely I bought a new board E3 Free-runs TMC2209 32bit CR4NT220801C10, I found on github the official firmware that came in the instructions for the board but it does not work for ender 5, it was made for ender 3, my question is can someone who is good at programming make me a firmware for ender 5/5 Pro?