CrealityOfficial/E3-Free-runs-Silent-Motherboard #8: Unable to update firmware

dannegomez, 2023-07-02:
Changed my Ender 3 Pro to the E3 board yesterday, compiled a new version with Marlin and BLTouch enabled. Copied the bin file to an SD card and stuck it into the printer, then turned the power on. However, the printer booted up without upgrading the firmware.

- I have tried renaming the file to all capital letters, all lowercase letters, and even using some random name, but nothing seems to get the printer to update the firmware.
- I have also tried using both an 8GB and 16GB SD card (both with FAT32), with the .bin file being the only file on it.

**Description**
The board does not update to the new firmware when booting with an SD card containing a .bin file.

**Expected behavior**
The board should flash with the "firmware.bin" file and enable BLTouch. The bed leveling menu should be visible on the LCD.

**Actual behavior**
The board boots up but does NOT flash the firmware or enable BLTouch.

Am I doing something wrong, or is there a problem with the E3 board?

ElasticSky, 2023-07-06:
Hey there,

Need to create a folder called "STM32F4_UPDATE" and then put the .bin file there. 

NOTE: The firmware file name needs to be different from the previous file name as the printer remembers the last name.

Kind regards

dannegomez, 2023-07-08:
Thank you @ElasticSky, that made it work

Curious to know, where did you find this information?

closed 2023-07-08 by dannegomez