CrealityOfficial/E3-Free-runs-Silent-Motherboard #18: Here is a firmware version for the cr touch for those who want it not sure if it will still work after being unzipped I haven't checked

BERKYNSON, 2024-02-24:
Here is a firmware version for the cr touch for those who want it not sure if it will still work after being unzipped I haven't checked
 don't forget to but the bin file into a folder on the sd card called "STM32F4_UPDATE" so it will flash the printer 
 
 FOR ENDER 3 please don't try and use this on a different printer :)

[firmwhere.zip](https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/files/12436207/firmwhere.zip)

_Originally posted by @The-sax-man in https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/issues/4#issuecomment-1692827803_

BERKYNSON, 2024-02-24:
Hello,

   The installation was a success with the simple opening from the button, I didn't even get to see well how nothing is shown on the display anymore. 3 or 4 sec. A very nice interface, I like the PID integration! 
   Unlike those from Creality, you seem to have managed to pleasantly surprise many people by making this version.
   The problem is that my enders 3 pro, don't have bltouch, and the prints fail after heating up. I need them to be fast, as close as possible to me, and I'd never had to recalibrate the bed more than 2 or 3 times per month for special prints.
   Can you please bless us with that 235x235x250 volume, and maybe about 300*C, with a version from which to remove the bltouch?
   I personally would be very grateful! I look forward to your feedback!

 All the best!
C. Brk.

The-sax-man, 2024-02-24:
I would love to fix that or to do what you ask here unfortunately it has been a very long time since I made the firmware and I have absolutely no clue how I had done it before, I would recommend trying to make your own firmware if you could I think it took me a few days to do it with lots of googling random stuff to make this version. I wish you luck if you decide to and I apologize I could not be more helpful.

donruan34, 2024-03-28:
Until I was able to install bltouch with the file you sent I was very happy but when I go to do a home it gives error m999 I saw that the z doesn't go down but you can see that...